NSEE/LPKF #10: Henrique Rosa - 11.02741-0 - 25/05/2016 - Completa

HenriquePereiraRosa, 2016-05-09:
# Sobre :

- Nome do Aluno : Henrique Pereira Rosa
- R.A : 11.02741-0
- Habilitação : Engenharia Eletrônica
- Ano : 2016
- Nome do Orientador: Rafafel Corsi 
- Nome do Projeto : Controle BLDC

# PCB :

- Nome da Placa : BLDC.PcbDoc
- Link para arquivos :  https://drive.google.com/open?id=0B0FWPGMv9ogrRzNIeXA3UVNWeG8
 (Pasta GERBERS controleBLDC v1.0)

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - [x] Altium
 - [ ] Eagle
 - [ ] Orcad
 - [ ] UltiBoard
 - [ ] Outro

- Tipo de prototipagem
 - [ ] Simples
 - [x] Completa

- Descrição :
 - [ ] Face Simples
 - [x] Dupla Face
 - [x] SMD
 - [ ] Trilha RF
 - [x] Máscara

EvertonFracaroli, 2016-05-10:
Boa Tarde @HenriquePereiraRosa, você poderia indicar o caminho exato dos arquivos.

EvertonFracaroli, 2016-05-10:
autorizado pelo Corsi via telefone

HenriquePereiraRosa, 2016-05-10:
Boa terde @EvertonFracaroli.

Corrigido: https://drive.google.com/open?id=0B0FWPGMv9ogrX3NvRWdKZXpQTUU

EvertonFracaroli, 2016-05-11:
Boa Tarde @HenriquePereiraRosa , vou ver com o @corsiferrao as extensões do seu arquivo, aqui no meu software esta dando alguns conflitos.

HenriquePereiraRosa, 2016-05-11:
OK! Qualquer coisa eu posso gerá-los junto com você aí na Mauá. Porque várias opções de exportação no Altium e as vezes é necessária alguma específica.

Qualquer novidade estou à disposição.

HenriquePereiraRosa, 2016-05-13:
Bom dia @EvertonFracaroli!

Abri o no Gerbv e realmente erro, porém em "Files >> Open layer(s)..." apresenta erro mas abre.
De qualquer gerei novamente os gerbers mesmo eles já estando de acordo com o guia Micropress.

Qualquer novidade é só me avisar.

EvertonFracaroli, 2016-05-13:
Obrigado @HenriquePereiraRosa , vou verificar.

EvertonFracaroli, 2016-05-13:
@HenriquePereiraRosa  e @corsiferrao, aparentemente esta tudo certo, agora vou fazer a placa e ver como vai ficar.

Obrigado

corsiferrao, 2016-05-17:
@HenriquePereiraRosa  estamos corrigindo um problema na câmera de vídeo da LPKF que é utilizada no processo de fabricação de PCBs dupla face, pode ser que não conseguimos entregar no prazo especificado. 

Mantemos vocês informados.

closed 2016-10-04 by corsiferrao