NSEE/LPKF #8: Placa Teste

EvertonFracaroli, 2016-05-09:
# Sobre : 

- Nome do Aluno : Everton
- R.A : 01.010101.01
- Habilitação :  Elétrica 
- Ano : 01
- Nome do Orientador: Rafael Corsi
- Nome do Projeto : Placa teste

# PCB :

- Nome da Placa : Placa Teste
- Link para arquivos : www.uol.com.br

## Informações básicas

- Pequeno descritivo sobre a placa : placa teste

- Software utilizado 
 - [ ] Altium
 - [ ] Eagle
 - [ ] Orcad
 - [x ] UltiBoard
 - [ ] Outro

- Tipo de prototipagem
 - [ x] Simples
 - [ ] Completa

- Descrição :
 - [ ] Face Simples
 - [ x] Dupla Face
 - [ ] SMD
 - [x ] Trilha RF
 - [ x] Máscara

EvertonFracaroli, 2016-05-09:
placa teste

corsiferrao, 2016-05-09:
@EvertonFracaroli título fora do padrão

corsiferrao, 2016-05-10:
@EvertonFracaroli 

- [ ] teste

closed 2016-05-11 by corsiferrao