NSEE/LPKF #9: Henrique Rosa - 11.02741-0 - 11/05/2016 - Simples

HenriquePereiraRosa, 2016-05-09:
# Sobre :

- Nome do Aluno : Henrique Pereira Rosa
- R.A : 11.02741-0
- Habilitação : Engenharia eletrônica
- Ano : 2016
- Nome do Orientador: Rafael Corsi 
- Nome do Projeto : Controle BLDC

# PCB : 

- Nome da Placa : BLDC.PcbDoc
- Link para arquivos : https://drive.google.com/open?id=0B0FWPGMv9ogrRzNIeXA3UVNWeG8
                  (Abrir pasta GERBERs)

## Informações básicas

- Pequeno descritivo sobre a placa : Controle de velocidade BLDC

- Software utilizado 
 - [x] Altium

- Tipo de prototipagem
 - [x] Simples

- Descrição :
 - [x] Dupla Face
 - [x] SMD
 - [x] Máscara

EvertonFracaroli, 2016-05-10:
Boa Tarde @HenriquePereiraRosa, vou poderia indicar o caminho exato dos arquivos.

HenriquePereiraRosa, 2016-05-10:
Segue o link: https://drive.google.com/open?id=0B0FWPGMv9ogrX3NvRWdKZXpQTUU

EvertonFracaroli, 2016-05-11:
@HenriquePereiraRosa o link https://drive.google.com/open?id=0B0FWPGMv9ogrX3NvRWdKZXpQTUUXpQTUU esta dando erro.

HenriquePereiraRosa, 2016-05-11:
@EvertonFracaroli estou atualizando o link. Mas esqueci de comentar que essa placa simples decidimos não fazer para economizar material. então será feita apenas a completa, tudo bem?

EvertonFracaroli, 2016-05-11:
@HenriquePereiraRosa  então podemos finalizar esse assunto e finalizar esse chamado?

HenriquePereiraRosa, 2016-05-11:
@EvertonFracaroli.
Sim, dessa placa simples sim! 
Muito obrigado.

closed 2016-05-11 by EvertonFracaroli