NSEE/LPKF #11: Carlos Eduardo Branco de Oliveira - 15.02419-9 - 11/05/2016 - Simples

cadubo, 2016-05-11:
# Sobre :

- Nome do Aluno : Carlos Eduardo Branco de Oliveira
- R.A : 15.02419-9
- Habilitação : Engenharia da Computação
- Ano : 2016
- Nome do Orientador: José Carlos 
- Nome do Projeto : Dado Digital
- Nome da Disciplina  : Fundamentos de Circuitos Digitais

# PCB :

- Nome da Placa : Dado Digital Projeto Final
- Link para arquivos: https://www.dropbox.com/sh/j3skv2r3ro61am3/AAD_012BOCplHKFj5h28y3Mwa?dl=0

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - (X) Outro

- Tipo de prototipagem
 - (X) Simples
 - ( ) Completa

- Descrição :
 - (X) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - (X) Máscara

EvertonFracaroli, 2016-05-11:
boa tarde Carlo, o que seria a máscara? 

Att,

Everton

corsiferrao, 2016-05-11:
@cadubo máscara deve-se entender aquela camada na PCB que a torna verde/preta/azul (como no caso do arduino), no seu caso acho que não tem máscara. 

Só realizamos máscara na prototipagem completa e para projetos de TCC, ic ...

cadubo, 2016-05-11:
boa noite everton,
Pensei que a máscara fosse outra coisa, é sem máscara mesmo

EvertonFracaroli, 2016-05-16:
@cadubo a placa esta pronta.

closed 2016-05-16 by EvertonFracaroli