NSEE/LPKF #13: Ronaldo Macarrao - 15.01715-0 - 16/05/2016 - Simples

ronaldomacarrao, 2016-05-13:
# Sobre :

- Nome do Aluno : Ronaldo Macarrao
- R.A :15.01715-0
- Habilitação :Engenharia Eletronica  
- Ano :2º
- Nome do Orientador:Marcio Mathias
- Nome do Projeto : FieldMill
- Nome da Disciplina  :PAE

# PCB :

- Nome da Placa : Fonte fieldmill
- Link para arquivos :https://www.dropbox.com/s/38ekokxdfdfuaiz/fieldmill%20fonte.zip?dl=0

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( x) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (x ) Simples
 - ( ) Completa

- Descrição :
 - (x ) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

corsiferrao, 2016-05-13:
@EvertonFracaroli abri exceção para esse pedido que foi enviado depois da data limite.
obrigado.

EvertonFracaroli, 2016-05-13:
@corsiferrao ok

EvertonFracaroli, 2016-05-16:
@ronaldomacarrao a placa esta pronta

closed 2016-05-16 by EvertonFracaroli