NSEE/LPKF #14: Guilherme Vicente Frizzo - 11.02555-7 - 11/05/2016 - Simples

gfrizzo, 2016-05-14:
# Sobre :

- Nome do Aluno : Guilherme Vicente Frizzo
- R.A : 11.02555-7
- Habilitação : CA
- Ano : 5o
- Nome do Orientador: Sergio Ribeiro
- Nome do Projeto : Telemetria embarcada
- Nome da Disciplina  : TCC

# PCB :

- Nome da Placa : Placa
- Link para arquivos : https://drive.google.com/folderview?id=0B_IU3fn9O9CXbmd0eDZzWDJ5clU&usp=sharing

## Informações básicas

- Pequeno descritivo sobre a placa : Shield para Arduino Mega cujo objetivo é comportar todos os sensores da telemetria

- Software utilizado 
 - ( ) Altium
 - ( X) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (X ) Simples
 - ( ) Completa

- Descrição :
 - ( ) Face Simples
 - (X ) Dupla Face
 - ( X) SMD
 - ( ) Trilha RF
 - (X ) Máscara

EvertonFracaroli, 2016-05-16:
@gfrizzo esta faltando o arquivo de borda. fico no aguardo.

gfrizzo, 2016-05-16:
Qual seria a extensão do arquivo?<div>
</div><div>
</div><div>-------- Mensagem original --------</div><div>De: EvertonFracaroli <notifications@github.com> </div><div>Data: 16/05/2016  10:03  (GMT-03:00) </div><div>Para: NSEE/LPKF <LPKF@noreply.github.com> </div><div>Cc: gfrizzo <gvfrizzo@hotmail.com>, Mention <mention@noreply.github.com> </div><div>Assunto: Re: [NSEE/LPKF] Guilherme Vicente Frizzo - 11.02555-7 - 16/05/2016 - Simples (#14) </div><div>
</div>
@gfrizzo esta faltando o arquivo de borda. fico no aguardo.

---
https://github.com/NSEE/LPKF/issues/14#issuecomment-219419629

EvertonFracaroli, 2016-05-16:
@gfrizzo alguns alunos entregaram com a extensão .OTL, mas nem todos ficou com o tamanho correto.

gfrizzo, 2016-05-16:
@EvertonFracaroli quando realizei o export não gerou a extensão .OTL. No entanto, olhando pelo Gerbv o contorno esta presente no arquivo de extensão .GTL. Poderia verificar? Vou buscar como gerar o arquivo OTL.

gfrizzo, 2016-05-17:
Upload feito para a pasta. Espero que seja suficiente e que fique bom.
Segui o tutorial deste link: http://optics.eee.nottingham.ac.uk/eagle/eagle2lpkf.html

corsiferrao, 2016-05-17:
@gfrizzo estamos corrigindo um problema na câmera de vídeo da LPKF que é utilizada no processo de fabricação de PCBs dupla face, pode ser que não conseguimos entregar no prazo especificado. 

Mantemos vocês informados.

EvertonFracaroli, 2016-05-17:
@gfrizzo agora deu certo com o arquivo placa.otl.

gfrizzo, 2016-05-23:
@corsiferrao alguma novidade?

corsiferrao, 2016-05-24:
@gfrizzo conseguimos arrumar o problema, devemos usinar sua placa ainda essa semana.

EvertonFracaroli, 2016-05-31:
@gfrizzo vc poderia me procurar, tenho uma duvida na sua placa. obrigado

gfrizzo, 2016-05-31:
Te procurarei amanhã de manhã @evertonfracaroli<div>
</div><div>
</div><div>-------- Mensagem original --------</div><div>De: EvertonFracaroli <notifications@github.com> </div><div>Data: 31/05/2016  08:05  (GMT-03:00) </div><div>Para: NSEE/LPKF <LPKF@noreply.github.com> </div><div>Cc: gfrizzo <gvfrizzo@hotmail.com>, Mention <mention@noreply.github.com> </div><div>Assunto: Re: [NSEE/LPKF] Guilherme Vicente Frizzo - 11.02555-7 - 16/05/2016 - Simples (#14) </div><div>
</div>
@gfrizzo vc poderia me procurar, tenho uma duvida na sua placa. obrigado

---
https://github.com/NSEE/LPKF/issues/14#issuecomment-222657706

EvertonFracaroli, 2016-05-31:
@gfrizzo o Sergio falou que vc esta querendo logo a placa, ela esta pronta só falta cortar porem tenho algumas duvidas, mas se vc  quiser já posso cortar e vc só retira ela.

gfrizzo, 2016-05-31:
@evertonfracaroli Dependendo do tempo que levar pra cortar eu passo ai antes e dps vou pra reunião com o prof Sergio.Não sei exatamente o problema então não sei afirmar qual a opção melhor<div>
</div><div>
</div><div>-------- Mensagem original --------</div><div>De: EvertonFracaroli <notifications@github.com> </div><div>Data: 31/05/2016  09:36  (GMT-03:00) </div><div>Para: NSEE/LPKF <LPKF@noreply.github.com> </div><div>Cc: gfrizzo <gvfrizzo@hotmail.com>, Mention <mention@noreply.github.com> </div><div>Assunto: Re: [NSEE/LPKF] Guilherme Vicente Frizzo - 11.02555-7 - 16/05/2016 - Simples (#14) </div><div>
</div>
@gfrizzo o Sergio falou que vc esta querendo logo a placa, ela esta pronta só falta cortar porem tenho algumas duvidas, mas se vc  quiser já posso cortar e vc só retira ela.

---
https://github.com/NSEE/LPKF/issues/14#issuecomment-222675367

gfrizzo, 2016-06-06:
Boa tarde @evertonfracaroli poderia por favor deixa a placa com alguem do almoxerifado ou com o professor Sergio? Irei retirar com um dos dois hoje a noite para analisar a inconsistência do resultado.Obrigado

-------- Mensagem original --------
De: Guilherme Frizzo <gvfrizzo@hotmail.com> 
Data: 31/05/2016 13:04 (GMT-03:00) 
Para: NSEE/LPKF <reply@reply.github.com>

Assunto: Re: [NSEE/LPKF] Guilherme Vicente Frizzo - 11.02555-7 - 16/05/2016 - Simples (#14)

@evertonfracaroli Dependendo do tempo que levar pra cortar eu passo ai antes e dps vou pra reunião com o prof Sergio.
Não sei exatamente o problema então não sei afirmar qual a opção melhor

-------- Mensagem original --------

view it on GitHub, or 
mute the thread.

EvertonFracaroli, 2016-06-06:
@gfrizzo  e só retirar a placa.

closed 2016-06-06 by EvertonFracaroli