NSEE/LPKF #15: MaurícioSchalch - Frequencímetro

schalchmauricio, 2016-05-16:
# Sobre :

- Nome do Aluno : Maurício Schalch
- R.A : 14.01112-3
- Habilitação : Engenharia Elétrica
- Ano : 3ºAno
- Nome do Orientador: Alessandra Dutra Coelho
- Nome do Projeto : Frequencímetro Digital
- Nome da Disciplina  : Eletrônica Digital/Analógica

# PCB :

- Nome da Placa : Frequencímetro
- Link para arquivos :  https://drive.google.com/folderview?id=0BzQjUqQw88D8cGoxX0JZWmp2VDg&usp=sharing

## Informações básicas

- Pequeno descritivo sobre a placa : 
O objetivo é prototipar a placa para ganhar tempo na execução do projeto solicitado no semestre.

Obs: Os arquivos gerados estão todos no formato .TXT devido à limitação do software usado. Foi testado o projeto no GerbV e a leitura foi possível, acredito que a LPKF lerá da mesma maneira.

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - (x) Outro (Proteus)

- Tipo de prototipagem
 - (x) Simples
 - ( ) Completa

- Descrição :
 - (x) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

corsiferrao, 2016-05-17:
@schalchmauricio o título deve manter o padrão especificado no tutorial, se atentar para isso da próxima vez.

EvertonFracaroli, 2016-05-18:
@schalchmauricio os arquivos estão certos.

schalchmauricio, 2016-05-19:
@EvertonFracaroli atualizei meu projeto para uma placa de menor dimensão.
Vou atualizar os arquivos de Gerber da nuvem (Google Drive).
Poderia confirmar se tem problema fazer isso?

EvertonFracaroli, 2016-05-19:
@schalchmauricio assim que vc alterar os arquivos vc me informa para verificar.

schalchmauricio, 2016-05-19:
Estao alterados ja

EvertonFracaroli, 2016-05-23:
@schalchmauricio os arquivos novos estão ok

EvertonFracaroli, 2016-05-27:
@schalchmauricio, a sua placa esta pronta.

closed 2016-05-27 by EvertonFracaroli