NSEE/LPKF #17: Gabriel Pianez - 14.00767-3 -30/05/2016- Simples

Bieldp3, 2016-05-19:
# Sobre :

- Nome do Aluno : Gabriel Pianez
- R.A : 14.00767-3
- Habilitação : Elétrica
- Ano :  3º
- Nome do Orientador:  Alessandra
- Nome do Projeto :  Frequencímetro
- Nome da Disciplina  : Eletrônica Digital

# PCB :

- Nome da Placa :projeto_final_alterado_1missing
- Link para arquivos : https://drive.google.com/file/d/0ByyJoplfhdclcjZjdjF0SXQ0V3M/view?usp=sharing

## Informações básicas

- Pequeno descritivo sobre a placa : O objetivo é prototipar a placa para ganhar tempo na execução do projeto solicitado no semestre.

Obs: Os arquivos gerados estão todos no formato .TXT devido à limitação do software usado. Foi testado o projeto no GerbV e a leitura foi possível, acredito que a LPKF lerá da mesma maneira.

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - (X ) Outro (Proteus)

- Tipo de prototipagem
 - (X ) Simples
 - ( ) Completa

- Descrição :
 - (X ) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2016-05-23:
@Bieldp3 os arquivos estão ok

EvertonFracaroli, 2016-05-27:
@Bieldp3 , a sua placa esta pronta.

closed 2016-05-27 by EvertonFracaroli