NSEE/LPKF #19: Bruno M. - 15.00237-3 - 20/06/2016 - Simples

bmarinotti, 2016-06-15:
# Sobre :

- Nome do Aluno : Bruno Marinotti Brostoline
- R.A : 15.00237-3
- Habilitação : Engenharia Elétrica
- Ano : 2ºAno
- Nome do Orientador: José Carlos
- Nome do Projeto : "Dado Digital"
- Nome da Disciplina  : Fundamentos de circuitos digitais

# PCB :

- Nome da Placa :  Dado Digital
- Link para arquivos :  https://drive.google.com/drive/folders/0B9O7RxjbnA9QTS13VTdoVGFTY1k

## Informações básicas

- Pequeno descritivo sobre a placa : Um contador de 00 a 99, que inicia-se com 00, e quando aperta-se o botão "Sortear" ele começa a contagem a 1kHz, após outro aperto no botão ele para a contagem e os seus display de 7 segmentos continuam mostrando o numero que parou a contagem, caso os dois digitos sejam iguais, uma lâmpada acenderá.

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - (x) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (x) Simples
 - () Completa

- Descrição :
 - ( ) Face Simples
 - (x) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - (x) Máscara
 - (x) Metalização

corsiferrao, 2016-06-16:
@EvertonFracaroli considerar o pedido do @bmarinotti para a placa ser feita fora do cronograma, você acha viável ?

corsi

EvertonFracaroli, 2016-06-17:
@corsiferrao sim e viável, porem acho que o custo de uma única placa com metalização e mascara e inviável, lembrando que depois que os produtos são abertos não tem como reaproveitar.

EvertonFracaroli, 2016-06-20:
@corsiferrao , O Prof. José Carlos autorizou a fazer a placa metalizada, mas sem a Mascara, O Prof. José Carlos orientou a fazer na Quinta-Feira e assim ver se entra mais uma placa para fazer.

corsiferrao, 2016-06-21:
@EvertonFracaroli ciente

EvertonFracaroli, 2016-06-21:
@bmarinotti esta faltando o Arquivo Board Outline.

bmarinotti, 2016-06-21:
Ok, obrigado por avisar, ja faço upload do gerber faltando no link

bmarinotti, 2016-06-22:
Todos os arquivos já foram atualizados e também adicionei o Board Outline.

EvertonFracaroli, 2016-06-23:
@bmarinotti os arquivos estão OK, assim que a placa estiver pronta aviso pelo GitHub.

EvertonFracaroli, 2016-06-23:
@bmarinotti sua placa estra pronta.

closed 2016-06-23 by EvertonFracaroli