NSEE/LPKF #25: Placa Driver Inversor de Freqüência

guibarsottini, 2016-10-14:
# Sobre :

- Nome do Aluno : Guilherme Barsottini Martines
- R.A :11.01772-4
- Habilitação : Enegenharia Eletrônica
- Ano : 5º
- Nome do Orientador: Alessandro Santos
- Nome do Projeto : Inversor de Freqüência 
- Nome da Disciplina  : EEE245 - Eletronica de Potencia

# PCB :

- Nome da Placa : Driver
- Link para arquivos : https://drive.google.com/open?id=0B7I-QIFYgvn8MGJNSVdpQ0w0Y0U

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - (X) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (X) Simples
 - ( ) Completa

- Descrição :
 - (X) Face Simples
 - ( ) Dupla Face
 - (X) SMD
 - ( ) Trilha RF
 - ( ) Máscara

corsiferrao, 2016-10-14:
[11:08, 13/10/2016] +55 11 99708-9707: Caro Corsi, bom dia...                         
[11:11, 13/10/2016] +55 11 99708-9707: Atendendo a sua solicitação, informo que o grupo do Srs. Barsotini e Zezel estam autorizados a pleitear um novo projeto de PCB na LPKF.                         
[11:11, 13/10/2016] +55 11 99708-9707: Um abraço... Prof. Alessandro

corsiferrao, 2016-10-14:
@EvertonFracaroli 
[08:59, 14/10/2016] +55 11 99193-1614: Bom dia ptofessor                         
[08:59, 14/10/2016] +55 11 99193-1614: Ontem 1 da manhã postei no git                         
[08:59, 14/10/2016] +55 11 99193-1614: Hahaha                         
[08:59, 14/10/2016] +55 11 99193-1614: Está dupla face mas se não estiver funcionando basta o botton layer                         
[09:00, 14/10/2016] +55 11 99193-1614: O resto será por Cabo mesmo                         
[09:00, 14/10/2016] +55 11 99193-1614: Desde ja agradeço a ajuda                         
[09:00, 14/10/2016] +55 11 99193-1614: Tenha um bom dia

corsiferrao, 2016-10-14:
@EvertonFracaroli se o problema com a dupla face ainda persistir, prototipar só o botton layer

EvertonFracaroli, 2016-10-18:
A placa vai ficar pronta no dia 20/10/16

EvertonFracaroli, 2016-10-20:
@guibarsottini a placa esta pronta. pode retirar na H227.

corsiferrao, 2016-10-20:
@guibarsottini  http://ieeexplore.ieee.org/Xplore/home.jsp

corsiferrao, 2016-10-20:
http://spectrum.ieee.org/

closed 2016-10-21 by EvertonFracaroli