NSEE/LPKF #27: Prototipagem

pedrohzo, 2016-10-20:
# Sobre :

- Nome do Aluno : Pedro Henrique Pontes Goncalves 
- R.A : 11.00226-3
- Habilitação : Eletronica
- Ano : 5 
- Nome do Orientador: Jose Carlos de Souza Jr.
- Nome do Projeto : Agripa
- Nome da Disciplina  : -

# PCB :

- Nome da Placa :Gerber Files Mux
[Gerber Files Mux.zip](https://github.com/NSEE/LPKF/files/540683/Gerber.Files.Mux.zip)

- Link para arquivos : https://drive.google.com/file/d/0B3n3t6DrrzUANTZXWFZCakJfNmc/view?usp=sharing

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - (X) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (X) Simples
 - ( ) Completa

- Descrição :
 - ( ) Face Simples
 - (X) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

pedrohzo, 2016-10-20:
Everton, eu mande dupla face pois nao consegui fazer em uma face simples. se houver problema me avise e ai converso contigo para mudar isso. Valeu!

EvertonFracaroli, 2016-10-20:
@pedrohzo  realmente a maquina não esta fazendo dupla face.

pedrohzo, 2016-10-20:
Minha placa possui 7 mux 8x1 e um Amp Op gostaria de saber se voce consegue me ajudar para fazer um layout de uma face so

EvertonFracaroli, 2016-10-21:
@pedrohzo nesse caso tem q fazer alguns jumper

pedrohzo, 2016-10-25:
@EvertonFracaroli bom dia Everton, atualizei o link do Gerber:
https://drive.google.com/open?id=0B3n3t6DrrzUAcThTbE9MUmFHT0E

Qualquer duvida me liga :972394529

EvertonFracaroli, 2016-10-28:
Pedro sua placa esta pronta

closed 2016-11-01 by EvertonFracaroli