NSEE/LPKF #28: Gimble

giu95, 2016-10-28:
# Sobre :

- Nome do Aluno : Giuliana Pasqualini
- R.A : 12.00188-0
- Habilitação : Engenharia de Controle e Automacao
- Ano : 2016
- Nome do Orientador: Vanderlei Parro
- Nome do Projeto : Gimble
- Nome da Disciplina  : Sistemas de Controle I

# PCB :

- Nome da Placa : Gimble
- Link para arquivos : https://drive.google.com/file/d/0B6Sm1ZQh6xu9TnZUUmNFM0NEcWc/view?usp=sharing

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - ( x ) Protheus

- Tipo de prototipagem
 - ( ) Simples
 - ( x ) Completa

- Descrição :
 - ( x ) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2016-10-31:
@giu95, os arquivos estão OK, porem a placa não tem plano de terra, nesses caso a gente não tira o excesso de cobre, isso não interfere em nada na placa e no seu funcionamento, mas se você achar melhor fazer com plano de terra e só avisar, sua placa esta na fila e vai ser feita amanha.

EvertonFracaroli, 2016-11-01:
@giu95 a placa esta pronta.

closed 2016-11-01 by EvertonFracaroli