NSEE/LPKF #29: Henrique Saboya - 14.00477-0 - 09/11/2016 - Simples

camelo1994, 2016-11-06:
# Sobre :

- Nome do Aluno : Henrique Miraglia Saboya
- R.A : 14.00477-0
- Habilitação : Controle e Automação - Diruno
- Ano : 3o
- Nome do Orientador: WANDERSON DE OLIVEIRA ASSIS - SERGIO RIBEIRO AUGUSTO
- Nome do Projeto : Amplificador com controle de volume digital
- Nome da Disciplina  : Eletronica Analógica/Digital (ETE202/ETE203)

# PCB :

- Nome da Placa : Amplificador Digital
- Link para arquivos : 
https://puu.sh/s8Kko/377f0bedf3.rar
https://drive.google.com/file/d/0B9E2s6Q6mc-kam9ER21PeVg4ZE0/view?usp=sharing

OBS: os arquivos gerber estão no formato TXT, pois foram gerados com o programa Proteus, todavia, o programa gerbv verificou os arquivos e todos estão corretos.

OBS2: no upload há um arquivo com extensao .RAR, com o conjunto dos arquivos.

OBS3: Caso nao seja possível imprimir a placa até a data alvo, nao realizar o pedido, uma vez que o trabalho tem que ser entregue até o dia seguinte, 10/11

## Informações básicas

- Pequeno descritivo sobre a placa :
Simples circuito para entrada de som de baixa amplitude, controle de amplitude com um comparador de tensao, controlado por um servo motor digitalmente, saida de até 30W, ajustado para 5W para segurança do falante usado nos testes.

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - ( X) Outro - Proteus

- Tipo de prototipagem
 - (X ) Simples
 - ( ) Completa

- Descrição :
 - (X ) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2016-11-07:
@camelo1994 notei que a sua placa não tem plano de terra, nesse caso a gente não retira o excesso de cobre que fica na placa.

EvertonFracaroli, 2016-11-08:
@camelo1994 a placa esta pronta.

closed 2016-11-16 by EvertonFracaroli