NSEE/LPKF #30: Matheus Antunes 14.03521-9 - 06/11/16 - Simples

mathantunes, 2016-11-06:
# Sobre :

- Nome do Aluno : Matheus Antunes
- R.A : 14.03521-9
- Habilitação : Engenharia Eletrônica
- Ano : 2016
- Nome do Orientador: 
- Nome do Projeto : Controle de Luminosidade
- Nome da Disciplina  :

# PCB :

- Nome da Placa : Lamp1
- Link para arquivos : https://drive.google.com/drive/folders/0B9EXYzDan5OWaXAtN2VTSW9xOVE?usp=sharing

## Informações básicas

- Pequeno descritivo sobre a placa : A partir de controle digital, um sensor de toque aciona uma lâmpada ligada na rede com intensidade controlada

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - (x ) Outro
 Proteus
- Tipo de prototipagem
 - (x) Simples
 - ( ) Completa

- Descrição :
 - ( ) Face Simples
 - (x) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2016-11-07:
@mathantunes notei que a sua placa e dupla face nesse caso tem que ter autorização, sua placa não tem plano de terra, nesse caso a gente não retira o excesso de cobre que fica na placa.

mathantunes, 2016-11-07:
@EvertonFracaroli Atualizei os arquivos, consegui colocar tudo na camada inferior e os limites da placa.

corsiferrao, 2016-11-08:
Ol'a @mathantunes você pode pedir para prototiparmos uma placa dupla face, a única questão é que para isso gastamos muito material e para minimizar desgastes das ferramentas da máquina e do uso de material para metalização utilizamos essa opção realmente quando necessário, mas para isso pedimos a autorização de algum professor responsável pelo projeto.

Verifique o que é melhor para o seu projeto, se quiser continuar na dupla face me passe o nome do Prof. que eu converso com ele para pedir a atuorização.

att,

Prof. R. Corsi

EvertonFracaroli, 2016-11-08:
@corsiferrao ele conseguiu fazer a placa em face simples, @mathantunes assim que a placa ficar pronta te aviso por esse canal.

EvertonFracaroli, 2016-11-08:
@mathantunes  a placa esta pronta.

closed 2016-11-16 by EvertonFracaroli