NSEE/LPKF #31: Amanda Fabri - 12028592 - 15/11/2016 - Simples

AmandaViviane, 2016-11-15:
# Sobre :

- Nome do Aluno : Amanda Viviane da Costa Fabri
- R.A : 12.02859-2
- Habilitação : Engenharia Eletrônica
- Ano : 4
- Nome do Orientador: Rafael Corsi
- Nome do Projeto : Chuveiro ecológico
- Nome da Disciplina  : EEN251 - Microcontroladores e Sistemas Embarcados

# PCB :

- Nome da Placa : Chuveiro ecológico
- Link para arquivos : https://drive.google.com/open?id=0B7z5zTwNT1RZVDFSTGRSVTZWWDA

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - ( x) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (x ) Simples
 - ( ) Completa

- Descrição :
 - ( x) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara 

[Gerber.zip](https://github.com/NSEE/LPKF/files/592257/Gerber.zip)

EvertonFracaroli, 2016-11-23:
@AmandaViviane a placa esta pronta, pode ser retirada na H227.

AmandaViviane, 2016-11-23:
Muito obrigada.

closed 2016-11-23 by EvertonFracaroli