NSEE/LPKF #33: Chuveiro Ecológico - 4ºEN - Noturno

FeMontagneri, 2016-11-28:
# Sobre :

- Nome do Aluno : Felipe Antonio Montagneri Lucchini
- R.A : 09,00053-4
- Habilitação :  Eneletrônica
- Ano : 4
- Nome do Orientador: José Carlos
- Nome do Projeto : Chuveiro Ecológico
- Nome da Disciplina  : Projetos em Eletrônica Aplicada

# PCB : 

- Nome da Placa :
- Link para arquivos : https://www.dropbox.com/s/v38rshz0un9e9lc/GERBER.rar?dl=0

## Informações básicas

- Pequeno descritivo sobre a placa :

E um protótipo para controle de potencia de um chuveiro.
Se possível retirar todo o cobre não utilizado, e utilizar a maior isolação que der.

- Software utilizado 
 - (x) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (x) Simples
 - ( ) Completa

- Descrição :
 - (x) Face Simples
 - ( ) Dupla Face
 - (x) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2016-11-29:
@FeMontagneri nesse caso por se tratar de uma segunda placa precisa de uma autorização do @corsiferrao.
@FeMontagneri  for informar o motivo pois e a mesma placa que foi feito o pedido da  Amanda Viviane da Costa Fabri R.A : 12.02859-2.

corsiferrao, 2016-11-29:
@EvertonFracaroli @FeMontagneri consultei o Prof. Julio e ele liberou a prototipação ,

corsi

FeMontagneri, 2016-11-29:
Obrigado!

Obter o Outlook para Android<https://aka.ms/ghei36>

EvertonFracaroli, 2016-11-29:
Placa já foi entregue pelo funcionário Renato.

closed 2016-11-29 by EvertonFracaroli