NSEE/LPKF #34: LoRa Radio

pedrohzo, 2016-12-07:
# Sobre :

- Nome do Aluno : Pedro
- R.A : 11.00226-3
- Habilitação : Eng. Eletrônica
- Ano :  5
- Nome do Orientador: Jose carlos de Souza Jr.
- Nome do Projeto : Radio Lora
- Nome da Disciplina  :

# PCB :

- Nome da Placa : Lora
- Link para arquivos : https://drive.google.com/open?id=0B3n3t6DrrzUAVFFQcGpvOTQ0a0k

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - ( X) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (X ) Simples
 - ( ) Completa

- Descrição :
 - ( ) Face Simples
 - ( X) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2016-12-12:
@pedrohzo tenho uma duvida, notei que tem 3 barramento no meio da placa, porém no aquivo de furação não tem furos para esse barramento é assim mesmo?

pedrohzo, 2016-12-12:
Vou tentar te mandar de novo. Teria que ter a furação dos barramentos sim. https://drive.google.com/open?id=0B3n3t6DrrzUAVFFQcGpvOTQ0a0k

EvertonFracaroli, 2016-12-13:
@pedrohzo  esse link já esta com os arquivos atualizados?

pedrohzo, 2016-12-16:
Sim esta

EvertonFracaroli, 2016-12-19:
Beleza, vou fazer ela hoje.

pedrohzo, 2016-12-19:
Valeu Everton! Qualquer coisa me manda mensagem

EvertonFracaroli, 2016-12-19:
@pedrohzo quando você consegue passar aqui na Mauá? a furação ainda esta incompleta.

pedrohzo, 2016-12-19:
Eu passo amanhã pode ser ?

EvertonFracaroli, 2016-12-20:
pode ser sim

pedrohzo, 2016-12-23:
https://drive.google.com/open?id=0B3n3t6DrrzUAakRCbzlDZGVOYzA

pedrohzo, 2016-12-23:
Atualizei e amanha o Rafael ira te ligar para conversar. Ve se este esta contendo as marcações! Valeu!!

EvertonFracaroli, 2017-01-04:
@pedrohzo o funcionário Renato esta de atestado, vou deixar a placa na porta da H227 com um durex, qualquer coisa me liga, 11 995202343.

Att

pedrohzo, 2017-01-07:
valeu Everton !

closed 2017-03-07 by EvertonFracaroli