NSEE/LPKF #37: Henrique Saboya - 14.00477-0 - 28/04/2017 - Face Simples

camelo1994, 2017-04-24:
# Sobre :

- Nome do Aluno : Henrique Miraglia Saboya
- R.A :14.00477-0
- Habilitação : Controle e Automação
- Ano : 4o
- Nome do Orientador: Fernando Silveira Madani
- Nome do Projeto : Placa genérica para prototipos com PIC
- Nome da Disciplina  : ECA203 - Microcontroladores e Microprocessadores

# PCB :

- Nome da Placa :pic40
- Link para arquivos :
 https://drive.google.com/file/d/0B9E2s6Q6mc-kcS0zT0pTNkEyTGc/view?usp=sharing
https://puu.sh/vuXSv/4f79b438ae.rar

## Informações básicas

- Pequeno descritivo sobre a placa : é como se fosse um arduino, mas com pic.

OBS: os gerbers gerados estão no formato *.txt, todavia foram verificados pelo software especificado.
OBS2: Se existir a possibilidade de imprimir top silkscreen na placa, para identificação dos pinos, seria bem interessante para o projeto.

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - (X) Outro - Proteus

- Tipo de prototipagem
 - (X) Simples
 - ( ) Completa

- Descrição :
 - (X) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - (X) Máscara  - se possível(?), devido a numeração dos pinos das portas

EvertonFracaroli, 2017-04-25:
Bom dia Henrique, vou esta verificando os arquivos hoje e já consigo adiantar que não temos os produtos para fazer o silkscreen.

EvertonFracaroli, 2017-04-27:
Boa Tarde Henrique, a placa esta pronta e pode ser retirada na Sala H227.

Att,

Everton

closed 2017-04-28 by EvertonFracaroli