NSEE/LPKF #43: Reeviando os arquivos de Caio Vinicius 13.02564-3

CaioVinicius100, 2017-05-15:
# Sobre :

- Nome do Aluno : Caio Vinicius Goncalves da Silva
- R.A : 13.02564-3
- Habilitação : CA
- Ano : 3
- Nome do Orientador: Alessandra/Wanderson
- Nome do Projeto :Novo Teste
- Nome da Disciplina  : Eletronica Analogica/Digital

# PCB :

- Nome da Placa : Finalzao
- Link para arquivos : https://drive.google.com/file/d/0B_btRdas7Siea3FOSzVpWHFTTVE/view?usp=sharing 

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( x) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (x ) Simples
 - ( ) Completa

- Descrição :
 - ( x) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2017-05-15:
Boa tarde Caio,

Sua placa ainda esta com furação de 0.2mm, quando vc alterar pode só mudar os arquivo no google drive e me avisar.

Att, 

Everton

CaioVinicius100, 2017-05-15:
Ola Everton, estou deixando aqui o Link para novo arquivo do projeto, eu revisei utilizando o software Gerbv e utilizei na Utilboard no Option -> Global Option -> Jumper size 0.6 mm, e alterou todos os jumpers para 0.6mm, e no software constatou o mesmo.

Link para arquivo ----------> https://drive.google.com/open?id=0B_btRdas7SieVzhsME5HWW1kbTQ

 Me avise que estou em aula, qualquer coisa eu posso conferir com voces.

EvertonFracaroli, 2017-05-16:
Caio ainda esta com furos de 0.2mm e sua placa esta fora do tamanho desejado que é de 8 x 10 cm a sua esta com 11 x 14 cm

EvertonFracaroli, 2017-05-16:
Caio a placa esta pronta

closed 2017-05-16 by EvertonFracaroli