NSEE/LPKF #46: Bruno Caldeira Fernandes RA 15.00730-8 15/05/17

brunodigitais, 2017-05-15:
# Sobre :

- Nome do Aluno : Bruno Caldeira Fernandes
- R.A : 15.00730-8
- Habilitação : Elétrica
- Ano : 3°
- Nome do Orientador: Wanderson
- Nome do Projeto : Projeto Semestral de Digitais
- Nome da Disciplina  : Eletrônica Digital

# PCB :

- Nome da Placa : Placa Digital
- Link para arquivos : https://www.dropbox.com/sh/ptdcbxk821ojqb0/AABZfJP_7-X8HL6KQVuMoRtDa?dl=0 

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - (X) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (X) Simples
 - ( ) Completa

- Descrição :
 - (X) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2017-05-16:
@brunodigitais , Sua placa tem furos de 0.2mm provavelmente seja dos jumpers, nesse caso utilizamos a broca de 0.6mm vc pode esta atualizando para essa broca e a placa esta fora do tamanho especificado o tamanho deve ser de 8 c 10 cmm. quando você atualizar os arquivos no dropbox você pode esta enviando uma mensagem pelo github.

Att,

Everton

brunodigitais, 2017-05-16:
@EvertonFracaroli Projeto arrumado, caso haja mais problemas por favor avisar.
Att,
Bruno

EvertonFracaroli, 2017-05-17:
Sua placa esta pronta, pode ser retirada.

closed 2017-05-17 by EvertonFracaroli