NSEE/LPKF #47: Leonardo Vila - 15.01232-8 - 15/05/2017 - Simples

LeoVila27, 2017-05-15:
# Sobre :

- Nome do Aluno : Leonardo Vila Franca Silva
- R.A : 15.01232-8
- Habilitação : Engenharia de Controle e Automação
- Ano : 3º
- Nome do Orientador:  Wanderson/Alessandra
- Nome do Projeto : Projeto Semestral
- Nome da Disciplina  : Eletrônica Analógica e Digital

# PCB :

- Nome da Placa : Tony
- Link para arquivos : https://drive.google.com/open?id=0BxWLKvrXzpxreW9fWmxHSlhhcHc

## Informações básicas

- Pequeno descritivo sobre a placa : Projeto de Eletrônica Digital/Analógica

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - (X ) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (X ) Simples
 - ( ) Completa

- Descrição :
 - (X ) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2017-05-16:
@LeoVila27  sua placa esta com o tamanho fora do padrão, o padrão e de 8 x 10 cm a sua esta com 14 x 11 cm. Você pode esta atualizando os arquivos no drive e me avisar pelo gitHub.

Att,

Everton

LeoVila27, 2017-05-16:
Obrigado por avisar Everton, os arquivos foram atualizados no drive -> https://drive.google.com/open?id=0BxWLKvrXzpxrVXNNSHhFYnJvRzg

Att,

Leonardo

EvertonFracaroli, 2017-05-17:
Sua placa esta pronta, pode ser retirada.

closed 2017-05-17 by EvertonFracaroli