NSEE/LPKF #48: Sulamita

Sulamitacl, 2017-05-15:
# Sobre :

- Nome do Aluno : Sulamita Carolina Lopes
- R.A : 15.03329-5
- Habilitação : Eletrica
- Ano : 3
- Nome do Orientador: Wanderson 
- Nome do Projeto : Sulinda
- Nome da Disciplina  : Eletrônica Digital

# PCB :

- Nome da Placa : Sulinda
- Link para arquivos : https://www.dropbox.com/sh/efhop86185tbyb4/AAAfYY-iB7gTB3MiuGHxpaYoa?dl=0

## Informações básicas

- Pequeno descritivo sobre a placa : 

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - (x) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (x) Simples
 - ( ) Completa

- Descrição :
 - (x ) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2017-05-16:
@Sulamitacl sua placa esta com o tamanho um pouco fora do padrão, ela ficou com 11 x 11 cm e nosso padrão e de 8 x10 cm acho que se você diminuir um pouco a borda você consegue chegar nessas medidas. 

Depois é só atualizar os arquivos no dropBox e me avisar pelo gitHub. 

Att,

Everton

EvertonFracaroli, 2017-05-16:
A placa esta pronta.

closed 2017-05-16 by EvertonFracaroli