NSEE/LPKF #49: Vinicius - 15.02077-0 - 15/05/2017 - Simples

ViniciusVTG, 2017-05-15:
# Sobre :

- Nome do Aluno :  Vinicius Vicente Targa Guadagnoli
- R.A : 15.02077-0
- Habilitação :  Controle e Automacão
- Ano :  3
- Nome do Orientador:  Wanderson/Alessandra
- Nome do Projeto : ProjetoSemestral
- Nome da Disciplina  : ETE202 - Eletronica Analogica e ECA305 - Eletronica Digital

# PCB :

- Nome da Placa : ProjetoSemestral
- Link para arquivos :  https://drive.google.com/open?id=0B_vLQY-SCpUrbUJ3RnN0R2NESU0

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( X) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - ( X) Simples
 - ( ) Completa

- Descrição :
 - ( X) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2017-05-16:
@ViniciusVTG Sua placa tem furos de 0.2mm provavelmente seja dos jumpers, nesse caso utilizamos a broca de 0.6mm vc pode esta atualizando para essa broca, quando você atualizar os arquivos no drive e você pode esta enviando uma mensagem pelo github.

Att,

Everton

ViniciusVTG, 2017-05-16:
@EvertonFracaroli Atualizei o arquivo. Agora deve estar tudo certo...

EvertonFracaroli, 2017-05-17:
Vinicius, comecei a fazer sua placa e vi que tem duas trilhas em curto, vc pode verificar, segue o linke com a foto. verifica se não tem mais assim.

https://drive.google.com/drive/folders/0BwoS62oMkZAQNWFTOEVRMTFQVGs

ViniciusVTG, 2017-05-17:
@EvertonFracaroli Arquivo atualizado.
Percebi um erro grave que deve ter causado o problema. Coloquei 0,5 na largura das trilhas mas esqueci de por em mm. Gerei uma nova placa e refiz as rotas. Agora deve estar tudo certo, apesar da placa estar um pouco maior (60x100).

EvertonFracaroli, 2017-05-22:
Bom dia a placa esta pronta

closed 2017-05-22 by EvertonFracaroli