NSEE/LPKF #50: Placa circuito impresso

psovich123, 2017-05-15:
# Sobre :

- Nome do Aluno :  Rafael Gass
- R.A : 15.03178-0
- Habilitação : controle e automação
- Ano : 3 ano
- Nome do Orientador:  Alessandra Dutra
- Nome do Projeto : projfreq PARA ENTREGAR na placa
- Nome da Disciplina  : Eletrônica Digital

# PCB :

- Nome da Placa : projfreq PARA ENTREGAR na placa
- Link para arquivos : https://www.dropbox.com/s/zeaotcv7m2tb7zg/projFreq%20PARA%20ENTREGAR%20na%20placa%20-%20CADCAM.ZIP?dl=0

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - (X) Outro

- Tipo de prototipagem
 - (X) Simples
 - ( ) Completa

- Descrição :
 - (X) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2017-05-16:
Sua placa tem furos de 0.4mm provavelmente seja dos jumpers, nesse caso utilizamos a broca de 0.6mm vc pode esta atualizando para essa broca e a placa esta fora do tamanho especificado o tamanho deve ser de 8 c 10 cmm, a sua placa esta com 23 x 22 cm, quando você atualizar os arquivos no dropbox você pode esta enviando uma mensagem pelo github.

Att,

Everton

leonwe, 2017-05-16:
Nome do Aluno : Rafael Gass
R.A : 15.03178-0
Habilitação : controle e automação
Ano : 3 ano
Nome do Orientador: Alessandra Dutra
Nome do Projeto : projfreq PARA ENTREGAR na placa
Nome da Disciplina : Eletrônica Digital
PCB :

Nome da Placa : projfreq PARA ENTREGAR na placa
Link para arquivos : https://www.dropbox.com/s/5g9l1neyfefto0u/projFreq%20PARA%20ENTREGAR%20na%20placa%20final%20-%20CADCAM.ZIP?dl=0
Informações básicas

Pequeno descritivo sobre a placa :

Software utilizado

( ) Altium

( ) Eagle

( ) Orcad

( ) UltiBoard

(X) Outro

Tipo de prototipagem

(X) Simples

( ) Completa

Descrição :

(X) Face Simples

( ) Dupla Face

( ) SMD

( ) Trilha RF

( ) Máscara

EvertonFracaroli, 2017-05-17:
Sua placa esta pronta, pode ser retirada.

closed 2017-05-17 by EvertonFracaroli