NSEE/LPKF #52: Projeto Semestral

nemerchiedde, 2017-05-15:
# Sobre :

- Nome do Aluno : Nemer Chiedde
- R.A : 15.03515-8
- Habilitação : Elétrica
- Ano : 3°ano
- Nome do Orientador: Wanderson e Alessandra
- Nome do Projeto:  Medidor de Frequência com Sensoriamento Indutivo
- Nome da Disciplina  :Eletrônica Analógica e Digital

# PCB :

- Nome da Placa : Projeto X
- Link para arquivos :

https://www.dropbox.com/sh/rcpg3j9d1bu5ta5/AAC9umAR0boXwQqkIbNHtKWua?dl=0

## Informações básicas

- Pequeno descritivo sobre a placa :
Projeto referente as disciplinas Eletrônica Analógica e Digital.

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( X ) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - ( X ) Simples
 - ( ) Completa

- Descrição :
 - ( X ) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2017-05-16:
Sua placa tem furos de 0.2mm provavelmente seja dos jumpers, nesse caso utilizamos a broca de 0.6mm vc pode esta atualizando para essa broca e a placa esta fora do tamanho especificado o tamanho deve ser de 8 c 10 cmm, a sua placa esta com 13 x 11 cm, quando você atualizar os arquivos no dropbox você pode esta enviando uma mensagem pelo github.

Att,

Everton

nemerchiedde, 2017-05-16:
Olá Everton,

Alterei o link com o arquivo corrigido.
Obrigado.

Nemer

EvertonFracaroli, 2017-05-17:
Sua placa esta pronta, pode ser retirada.

closed 2017-05-17 by EvertonFracaroli