NSEE/LPKF #53: Prototipagem

leonwe, 2017-05-15:
# Sobre :

- Nome do Aluno : Leon Cheroto Weissberg
- R.A : 13.01137-5
- Habilitação : CA
- Ano : 3º serie
- Nome do Orientador: Alessandra Dutra
- Nome do Projeto : projeto semestral 1
- Nome da Disciplina  : Eletrônica Digital

# PCB :

- Nome da Placa : Projeto semestral 1
- Link para arquivos : https://www.dropbox.com/s/4m27lvvtywpc0w1/projeto%20semetral%201%20-%20CADCAM.ZIP?dl=0

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - (X ) Outro

- Tipo de prototipagem
 - (X ) Simples
 - ( ) Completa

- Descrição :
 - ( X) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2017-05-16:
Sua placa tem furos de 0.4mm provavelmente seja dos jumpers, nesse caso utilizamos a broca de 0.6mm vc pode esta atualizando para essa broca e a placa esta fora do tamanho especificado o tamanho deve ser de 8 c 10 cmm, a sua placa esta com 23 x 22 cm, quando você atualizar os arquivos no dropbox você pode esta enviando uma mensagem pelo github.

Att,

Everton

leonwe, 2017-05-16:
Link atualizado

EvertonFracaroli, 2017-05-17:
@leonwe na hora que você gerou os arquivos deu alguma coisa errada, fiz um print da tela. segue o link

https://drive.google.com/drive/folders/0BwoS62oMkZAQYTluYmRDMlhuSlU

leonwe, 2017-05-17:
link atualizado

EvertonFracaroli, 2017-05-22:
Bom dia a placa esta pronta

closed 2017-05-22 by EvertonFracaroli