NSEE/LPKF #56: Pedro Nicolau Arcuri - 16.95021-6 - 18/05/2017 - Simples

pedronic, 2017-05-18:
# Sobre :

- Nome do Aluno : Pedro Nicolau Arcuri
- R.A : 16.95021-6
- Habilitação : Engenharia de Controle e Automação
- Ano : 3°
- Nome do Orientador: Alessandra Dutra
- Nome do Projeto : Projeto Semestral 1 - Frequencímetro Digital
- Nome da Disciplina : Eletrônica Analógica / Eletrônica Digital

# PCB :

- Nome da Placa : Leonardo_Andre_Pedro - S1 - PCB
- Link para arquivos : https://drive.google.com/drive/folders/0B8I5asxs4dz6WmotdERqMGp6bWc?usp=sharing

## Informações básicas

- Pequeno descritivo sobre a placa :   PCB para o Projeto Semestral integrado entre as disciplinas Eletrônica Analógica e Eletrônica Digital (referente ao primeiro semestre de 2017)

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - (×) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (×) Simples
 - ( ) Completa

- Descrição :
 - (×) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2017-05-26:
Pedro bom dia, cara sua placa não sei por qual motivo, mas quando vc gerou ela ficou com vários curtos entre as trilhas. so vi quando a LPKF terminou de fazer ela. tem como vc gerar novamente os arquivos. obrigado.

pedronic, 2017-05-28:
Bom dia, Everton! Modifiquei as trilhas do Copper Bottom que estavam muito próximas umas das outras, adicionei alguns jumpers e aparentemente não há nenhum CC. Atualizei os arquivos do Drive com os Gebers gerados. Grato!

EvertonFracaroli, 2017-05-29:
Verifiquei os arquivos e estão certos, se encontrar alguma coisa na hora de fazer a placa te aviso

EvertonFracaroli, 2017-05-30:
bom dia, sua placa esta pronta. pode ser retirada na H227

Obrigado

closed 2017-05-30 by EvertonFracaroli