NSEE/LPKF #57: Leonardo M. - 14.03930-3 - 25/5/2017 - Simples

leollacer, 2017-05-18:
# Sobre :

- Nome do Aluno : Leonardo Molino Llacer
- R.A : 14.03930-3
- Habilitação : Engenharia Controle e Automação
- Ano : 3º
- Nome do Orientador:  Alessandra Dutra
- Nome do Projeto : Frequencimetro
- Nome da Disciplina  : Eletrônica Digital

# PCB : 

- Nome da Placa : Leo_Vagner
- Link para arquivos :  https://1drv.ms/f/s!AswBuWH2XhxEiTNAL_xwBdTweS0z

## Informações básicas
Placa feita para o projeto de eletrônica analogia e eletrônica digital do terceiro ano de controle e automação.

- Pequeno descritivo sobre a placa :
 Placa feita para o projeto de eletrônica analogia e eletrônica digital do terceiro ano de controle e automação.
- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - (x) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (x) Simples
 - ( ) Completa

- Descrição :
 - (x) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2017-05-22:
Bom dia, 
Seu arquivo de borda esta com 25 x 31cmm e sua trilhas estão finas.

leollacer, 2017-05-22:
Segue link para correção:
thank you. 
https://1drv.ms/f/s!AswBuWH2XhxEiUYaw5V2nhiCtdd-

EvertonFracaroli, 2017-05-23:
@leollacer sua borda ainda esta muito grande, se estiver usando o ultiboard seleciona o layer boardoutline, clica na borda em amarelo e que ele vai dar a opção de ajustar a borda.

leollacer, 2017-05-23:
Olá Everton
O problema já foi corrigido, segue a correção.
Obrigado

EvertonFracaroli, 2017-05-26:
Bom dia, fiz a sua placa mas ela esta com as trilhas muito finas e muito próxima uma da outra, a maquina fez mas comeu algumas parte das trilha, vou pedir para você deixar mais espaços entre as trilhas e deixar as trilhas mias grosas.

leollacer, 2017-05-26:
Olá Everton
O problema já foi corrigido, segue a correção.
https://drive.google.com/open?id=0B2pZnbKu4_E6MGcyMzRtd1hIVVE
Obrigado

EvertonFracaroli, 2017-05-29:
Leonardo sua placa esta com os arquivos certos, porem vou pedir para você diminuir um pouco a borda da placa. Outra coisa notei que você não colocou entrada para alimentação da placa e as saídas do display é isso mesmo?

leollacer, 2017-05-29:
Everton, bom dia. Não existe a possibilidade de ser feito com esse tamanho de borda ? Pois só conseguimos modificar a noite e precisamos dela para soldar hj e mostrar ao professor se ta ok.

leollacer, 2017-05-30:
Olá Everton
O problema já foi corrigido, segue a correção.
https://drive.google.com/drive/folders/0B3CpxDT9jeGrOHVXSEtYX05ZOGc
E informando minha dupla vagner também te mandou a correção.

EvertonFracaroli, 2017-05-31:
Conversei com o renato e ele informou que já fez a entrega da placa.

closed 2017-05-31 by EvertonFracaroli