NSEE/LPKF #58: José Everlando Cordeiro Júnior - 14.01181-6 - 18/05/2017 - Simples

Josejunior96, 2017-05-18:
# Sobre :

- Nome do Aluno : José Everlando Cordeiro Júnior
- R.A : 14.01181-6
- Habilitação : Controle e Automação
- Ano : 3º
- Nome do Orientador: Alessandra Dutra.
- Nome do Projeto : Projeto Semestral 1 - Frequencímetro Digital
- Nome da Disciplina  : Eletrônica Analógica e Eletrônica Digital

# PCB :

- Nome da Placa : Jose_Ultiboard
- Link para arquivos : https://1drv.ms/f/s!AswBuWH2XhxEiT4hmvxOLzaI8mec

## Informações básicas

- Pequeno descritivo sobre a placa :  Projeto de PCB referente ao Projeto de eletrônica Analógica e Eletrônica digital do curso de Engenharia de Controle e Automação do 3º ano.

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - (x) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (x) Simples
 - ( ) Completa

- Descrição :
 - (x) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2017-05-22:
Bom dia, 

Você precisa gerar os gebers.

Josejunior96, 2017-05-22:
Boa noite, 
Segue link do arquivo atualizado sem erros de Gebers! 

Muito obrigado!

https://1drv.ms/f/s!AswBuWH2XhxEiUKI1QJvunY4VgIK

EvertonFracaroli, 2017-05-23:
@Josejunior96 esta faltando o arquivo de borda - **BoardOutline**

leollacer, 2017-05-23:
Boa tarde @EvertonFracaroli ,  segue link : https://1drv.ms/f/s!AswBuWH2XhxEiUr-zXjsfn2JX-uL 
Espero que dessa vez esteja certo! Obrigado.

Josejunior96, 2017-05-23:
@EvertonFracaroli , estava logado aqui na máquina como esse @leollacer, mas o arquivo que te enviei é meu @Josejunior96.

EvertonFracaroli, 2017-05-26:
Bom dia, fiz a sua placa mas ela esta com as trilhas muito finas e muito próxima uma da outra, a maquina fez mas comeu algumas parte das trilha, vou pedir para você deixar mais espaços entre as trilhas e deixar as trilhas mias grosas.

Josejunior96, 2017-05-26:
Bom dia @EvertonFracaroli, só vou conseguir alterar as trilhas umas 17h, que é o horário que eu chego na Mauá. De qualquer forma, agradeço!

Josejunior96, 2017-05-26:
Boa noite @EvertonFracarolim, acredito que finalizei o arquivo. Espero que dessa vez dê certo, obrigado! 
https://1drv.ms/f/s!AswBuWH2XhxEiVAbrEk4uFZ5eS85

EvertonFracaroli, 2017-05-29:
Leonardo sua placa esta com os arquivos certos, porem vou pedir para você diminuir um pouco a borda da placa. Outra coisa notei que você não colocou entrada para alimentação da placa e as saídas do display é isso mesmo?

Josejunior96, 2017-05-29:
Boa tarde @EvertonFracaroli , segue link atualizado https://1drv.ms/f/s!AswBuWH2XhxEiVag8JBZkGDQRP7r.   Apenas diminui as bordas, porém não alterar entrada, vamos soldar direto. Obrigado!

EvertonFracaroli, 2017-05-31:
Conversei com o Renato e ele informou que já fez a entrega da placa.

closed 2017-05-31 by EvertonFracaroli