NSEE/LPKF #59: Projeto Digital

gukobayashi, 2017-05-19:
# Sobre :

- Nome do Aluno : Gustavo Kobayashi Ramos
- R.A :15.01692-7
- Habilitação : Engenharia Elétrica
- Ano : 3
- Nome do Orientador: Alessandra Dutra
- Nome do Projeto : Projeto 14
- Nome da Disciplina  : Eletrônica Digital

# PCB :

- Nome da Placa : Frequencimetro
- Link para arquivos : https://www.dropbox.com/sh/y6asqslfd859h07/AADfLSAUP172DRPrHjpmaD6Wa?dl=0

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - (x) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (x) Simples
 - ( ) Completa

- Descrição :
 - (x) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2017-05-22:
bom dia,

Sua placa esta fora do tamanho padrão que é de 8x10cm e ela contem furos de 0.2mm sendo que a menor broca que temos e de 0.6mm

gukobayashi, 2017-05-24:
Agora sim, creio que esta tudo certo.

https://www.dropbox.com/sh/y6asqslfd859h07/AADfLSAUP172DRPrHjpmaD6Wa?dl=0

closed 2017-05-26 by EvertonFracaroli