NSEE/LPKF #60: Victor Loiola - 13.0254-4

Sulamitacl, 2017-05-25:
# Sobre :
- Nome do Aluno :  Victor Loiola
- R.A : 13.02054-4
- Habilitação : Elétrica
- Ano : 3°
- Nome do Orientador: Sérgio Ribeiro
- Nome do Projeto : Placa 2
- Nome da Disciplina  : Circuitos Digitais

# PCB :

- Nome da Placa : Placa 2
- Link para arquivos : 

https://www.dropbox.com/sh/jhhql5e2qa59xyn/AABFt_MyDaBzhf0CiHL9C0s1a?dl=0

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - (X) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (X) Simples
 - ( ) Completa

- Descrição :
 - (X) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2017-05-29:
Bom dia Victor, sua placa esta fora do tamanho padra que é de 8 x 10 cm e suas trilhas estão muitos finas. quando você arrumar, pode esta atualizando os arquivos no dropbox e me avisar por mensagem pelo github.

Obrigado

EvertonFracaroli, 2017-05-31:
bom dia, você conseguiu arrumar a placa?

closed 2017-06-01 by EvertonFracaroli