NSEE/LPKF #66: Vagner Possato Junior - 15.03236-0 - 30/5/2017 - Simples

VagnerPossato, 2017-05-30:
# Sobre :

- Nome do Aluno : Vagner Possato Junior
- R.A : 15.03236-0
- Habilitação : Controle e Automação
- Ano : 3º
- Nome do Orientador:  Alessandra Dutra.
- Nome do Projeto : Projeto Semestral 1 - Frequencímetro Digital
- Nome da Disciplina  : Eletrônica Analógica e Eletrônica Digital

# PCB :

- Nome da Placa : Vagner_Ultiboard
- Link para arquivos : https://drive.google.com/drive/folders/0B3CpxDT9jeGrOHVXSEtYX05ZOGc

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - (x) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (x) Simples
 - ( ) Completa

- Descrição :
 - (x) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

VagnerPossato, 2017-05-30:
Bom dia @EvertonFracaroli 
Segue link referente a correção da placa do Leonardo Molino, minha dupla: https://drive.google.com/drive/folders/0B3CpxDT9jeGrOHVXSEtYX05ZOGc
Agradeço pela atenção!

VagnerPossato, 2017-05-30:
Bom dia @EvertonFracaroli 
Segue correção da placa do Leonardo Molino minha dupla:
https://drive.google.com/drive/folders/0B3CpxDT9jeGrOHVXSEtYX05ZOGc

VagnerPossato, 2017-05-30:
Boa Tarde,
A respeito dos pinos de entrada e saída iremos soldar os fios na placa, pois a parte analógica foi montada separadamente.
Obrigado

EvertonFracaroli, 2017-05-31:
Conversei com o renato e ele informou que já fez a entrega da placa.

closed 2017-05-31 by EvertonFracaroli