NSEE/LPKF #70: Contador de Frequência

nicolasumaras, 2017-06-02:
# Sobre :

- Nome do Aluno : Nícolas Rubliauskas Umaras
- R.A : 14.02014-9
- Habilitação : Engenharia Eletrônica
- Ano : 3
- Nome do Orientador: Pousada
- Nome do Projeto : contador de frequências
- Nome da Disciplina  : Circuitos Analógicos/Digitais

# PCB :

- Nome da Placa : Contador de Frequencia
- Link para arquivos : https://drive.google.com/drive/folders/0B3VWuvLS71AQN2tnN3VCNmJYWE0?usp=sharing

## Informações básicas

- Pequeno descritivo sobre a placa : placa composta por CIs e alguns componentes, apenas camada inferior e jumpers na camada superior. 0,5mm de largura de trilha

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - (X) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (X) Simples
 - ( ) Completa

- Descrição :
 - (X) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2017-06-02:
Cara tem como vc passar na H227 para ver como vcs estão gerando os arquivos, ainda continua errado

EvertonFracaroli, 2017-06-05:
O arquivo continua errado

EvertonFracaroli, 2017-06-06:
A placa esta pronta

closed 2017-06-06 by EvertonFracaroli