NSEE/LPKF #72: Placa Trabalho Altas Frequências

ThalesPC, 2017-06-06:
# Sobre :

- Nome do Aluno : Thales Pizzotti Costi
- R.A :13.00946-0
- Habilitação : Eng. Eletrônica
- Ano : 5º Diurno
- Nome do Orientador: Marcio Mathias
- Nome do Projeto :Placa Trabalho Altas Frequências
- Nome da Disciplina  : EEN941 - Técnicas de Engenharia para Altas Frequências I

# PCB :

- Nome da Placa :
- Link para arquivos : https://www.dropbox.com/s/jhoorwp59m0hfy1/GERBERATUALIZADO.rar?dl=0

## Informações básicas

- Pequeno descritivo sobre a placa : A placa contém um acoplador direcional e um filtro passa baixas, desenvolvido nas técnicas de microfitas.
Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - (x ) Outro -PCAD

- Tipo de prototipagem
 - ( ) Simples
 - (x ) Completa

- Descrição :
 - ( ) Face Simples
 - (x ) Dupla Face (A face inferior não possui trilha, sendo apenas o plano de terra onde conectamos os conectores SMA)
 - (x ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2017-06-06:
A placa esta pronta, pode ser retirada na H227

ThalesPC, 2017-06-10:
Bom dia Everton,
Segue o link dos arquivos corrigidos:
https://drive.google.com/file/d/0B5xnmylZZ-SXVzV0RndPaFRZbjQ/view?usp=sharing
Grato.
Thales.

closed 2017-06-16 by EvertonFracaroli