NSEE/LPKF #74: Henrique Rosa - 11.02741-0 - 12/6/2017 - Simples

HenriquePereiraRosa, 2017-06-08:
# Sobre : Casador de impedância baseado em toco simples.

- Nome do Aluno : Henrique Pereira Rosa
- R.A : 11.02741-0
- Habilitação : Eletrônica
- Ano : 5º
- Nome do Orientador:  Marcio Mathias
- Nome do Projeto : Projeto final da disciplina
- Nome da Disciplina  : EEN941 - Técnicas em Altas Frequências I

# PCB :

- Nome da Placa : CasadorTocoParalelo.pcbDoc
- Link para arquivos :  https://drive.google.com/open?id=0B0FWPGMv9ogrLXJUUGh5MlhfQ2c

## Informações básicas

- Pequeno descritivo sobre a placa :
       Casador de impedância baseado em toco simples.

- Software utilizado 
 - (x) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (x) Simples
 - ( ) Completa

- Descrição :
 - ( ) Face Simples
 - (x) Dupla Face (A parte inferior é só um chapado de cobre). 
 - ( ) SMD
 - (x) Trilha RF
 - ( ) Máscara

HenriquePereiraRosa, 2017-06-27:
@EvertonFracaroli 
Oi Everton. Acho que você fechou esse achando que era a mesma placa, mas esse ISSUE é outra placa.

EvertonFracaroli, 2017-06-28:
Opa beleza vou ver aqui.

HenriquePereiraRosa, 2017-06-28:
Valeu!

EvertonFracaroli, 2017-06-28:
a placa esta pronta.

closed 2017-06-29 by EvertonFracaroli