NSEE/LPKF #75: Henrique Rosa - 11.02741-0 - 12/06/2017 - Simples

HenriquePereiraRosa, 2017-06-08:
# Sobre : Casador de impedância baseado em transformador multiseção (binomial).

- Nome do Aluno : Henrique Pereira Rosa
- R.A : 11.02741-0
- Habilitação : Eletrônica
- Ano : 5º
- Nome do Orientador:  Marcio Mathias
- Nome do Projeto : Projeto final da disciplina
- Nome da Disciplina  : EEN941 - Técnicas em Altas Frequências I

# PCB :
- Nome da Placa : CasadoMultisecaoBinomial.PcbDoc
- Link para arquivos :  https://drive.google.com/open?id=0B0FWPGMv9ogrX2tua0tPZjRaWHc

## Informações básicas

- Pequeno descritivo sobre a placa :
       Casador de impedância baseado em transformador multiseção (binomial).

- Software utilizado 
 - (x) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (x) Simples
 - ( ) Completa

- Descrição :
 - ( ) Face Simples
 - (x) Dupla Face (A parte inferior é só um chapado de cobre). 
 - ( ) SMD
 - (x) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2017-06-22:
@HenriquePereiraRosa Essa placa não tem furação mesmo?

HenriquePereiraRosa, 2017-06-23:
Isso! sem furação.

EvertonFracaroli, 2017-06-23:
ok

EvertonFracaroli, 2017-06-23:
A placa esta pronta

closed 2017-06-23 by EvertonFracaroli

HenriquePereiraRosa, 2017-06-23:
Valeu!!