NSEE/LPKF #78: TCC - Detector Falta de Fase

pedrohrocha, 2017-06-27:
# Sobre :

- Nome do Aluno : Pedro Henrique Rocha
- R.A : 11.00776-0
- Habilitação : Elétrica
- Ano : 2017
- Nome do Orientador: Antonio Saverio
- Nome do Projeto : Detector e Alerta de Falta de Fase
- Nome da Disciplina : TCC

# PCB :

- Nome da Placa : TCC - Detector de Falta de Fase
- Link para arquivos : https://www.dropbox.com/sh/2eumvf9pb4sppmh/AACkDnzzTZVKU38AdEVacUFba?dl=0

## Informações básicas

- Pequeno descritivo sobre a placa : Esta placa é responsável por detectar uma falha de alimentação de uma instalação, sendo em umas das 3 fases ou neutro, e gerar um sinal que será enviado para um sistema de comunicação que alertará o operador sobre esta falta de alimentação.

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - (X) Outro

- Tipo de prototipagem
 - (X) Simples
 - ( ) Completa

- Descrição :
 - (X) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2017-06-28:
@pedrohrocha a parte que esta sem plano de terra a gente não tem mais a broca para tirar o cobre.

pedrohrocha, 2017-06-28:
Da pra deixar com o cobre respeitando as trilhas?

EvertonFracaroli, 2017-06-29:
Sim ela vai fazer as trilhas normal, mas o cobra vai ficar, vi que vc fez plano de terra só em uma tarde da placa. outra duvida, a placa esta feita no lado TOP e isso mesmo?

pedrohrocha, 2017-06-29:
td bem o restante ficar com o cobre..
é mesmo..eu fiz na TOP.. mas acho q nao tem problema, eu soldo os componentes no outro lado. se der pra inverter e colocar no Bottom é melhor,

EvertonFracaroli, 2017-06-29:
E melhor não inverter pode dar alguma coisa errada

pedrohrocha, 2017-06-29:
entao faz assim mesmo. é só soldar os componentes no bottom..rs

pedrohrocha, 2017-06-29:
mas se quiser eu altero o projeto.

EvertonFracaroli, 2017-06-29:
você que sabe, para a gente aqui não tem problemas se é top ou bottom

pedrohrocha, 2017-06-29:
entao td bem..td bem..pode fazer no top

EvertonFracaroli, 2017-06-29:
ok

EvertonFracaroli, 2017-06-30:
a placa esta pronta

closed 2017-07-04 by EvertonFracaroli