NSEE/LPKF #79: Pedro R. - 11.00776-0 - 20/07/2017 - Simples

pedrohrocha, 2017-07-20:
# Sobre :

- Nome do Aluno : Pedro Henrique Rocha
- R.A : 11.00776-0
- Habilitação : Elétrica
- Ano : 6º noturno
- Nome do Orientador: Antonio Saverio
- Nome do Projeto : Detector e Alerta de Falta de Fase
- Nome da Disciplina  : TCC

# PCB :

- Nome da Placa :  TCC - Detector de Falta de Fase
- Link para arquivos : https://drive.google.com/drive/folders/0B8wwFdi9lWcgNnZmU2t3cWVPSXc

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - (X) Outro

- Tipo de prototipagem
 - (X) Simples
 - ( ) Completa

- Descrição :
 - (X) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

pedrohrocha, 2017-07-26:
Olá Everton. 
algum problema com os arquivos?
eu verifiquei pelo programa Gerbv e abriu normal.

EvertonFracaroli, 2017-07-26:
Vamos verificar os arquivos ainda

EvertonFracaroli, 2017-07-27:
A placa esta pronta

pedrohrocha, 2017-07-27:
Obrigado.. vou buscar hoje de noite.

closed 2017-07-27 by EvertonFracaroli