NSEE/LPKF #98: Diogo Slepetys - 13.01627-0 - 24/10/2017 - Simples

DiogoSlepetys, 2017-10-20:
# Sobre :

- Nome do Aluno : Diogo Slepetys
- R.A : 13.01627-0
- Habilitação : Computação
- Ano : 4º
- Nome do Orientador: Vanderlei Parro
- Nome do Projeto : Conversor de pitch p/ Módulo RF v2
- Nome da Disciplina : Iniciação Científica - NSEE

# PCB :

- Nome da Placa : Conversor de pitch p/ Módulo RF v2
- Link para arquivos : [https://drive.google.com/drive/folders/0B7YJ5ZX5FAZGcVJ0QVg4NEQ4VjA?usp=sharing](https://drive.google.com/drive/folders/0B7YJ5ZX5FAZGcVJ0QVg4NEQ4VjA?usp=sharing)

## Informações básicas

- Obs: Não será necessário furar a placa.

- Pequeno descritivo sobre a placa :

- Software utilizado 

 - ( x ) Eagle

- Tipo de prototipagem

 - ( x ) Simples

- Descrição :

 - ( x ) Face Simples

Renatolab, 2017-10-26:
Boa tarde Diogo.
Gostaria de saber se sua placa não vai furo mesmo?
Sua Placa é top Layer ou Bottom Layer?
Também está faltando o Board out line, que é a marcação ao redor da placa.
Aguardo contato
Obrigado
Renato

DiogoSlepetys, 2017-10-27:
Boa tarde Renato.

Não vou precisar dos furos.
A placa é top layer.

Já havia pedido uma placa igual a essa no [issue 90](https://github.com/NSEE/LPKF/issues/90) e foi feita com o board outline corretamente. Foi feita apenas uma pequena correção nas trilhas nessa versão.

Acredito que o outline esteja correto pois ele está presente quando utilizo visualizadores de gerber para verificar a integridade dos arquivos.

Os arquivos foram gerados a partir do job: gerber274x no CAM processor do Eagle, assim como para a placa anterior. Posso tentar alterar o formato caso necessário.

Muito obrigado,
Diogo

Renatolab, 2017-10-30:
Boa tarde Diogo.
Sua Placa está pronta.
Renato

closed 2017-10-31 by Renatolab