NSEE/LPKF #99: Felipe B. Ippolito - 12.01378-0 - 24/10/2017 - Simples

feippolito, 2017-10-22:
# Sobre :

- Nome do Aluno : Felipe Brandão Ippolito
- R.A :12.01378-0
- Habilitação : Eng. Eletrônica
- Ano : 2
- Nome do Orientador: Valdir Junior
- Nome do Projeto : Fonte Linear 15V
- Nome da Disciplina  : ET3001- Projetos Eletrônicos I - Fontes de Alimentação Lineares

# PCB :

- Nome da Placa : Fonte 15 V
- Link para arquivos : https://drive.google.com/open?id=0B5LCK_Lxpda9a084Vk51MXNlbXc

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - (X) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (X) Simples
 - ( ) Completa

- Descrição :
 - (X) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

EvertonFracaroli, 2017-10-24:
Sua placa esta com as trilhas muito fina a LPKF não consegue fazer.

feippolito, 2017-10-24:
Criei um novo Issue com as trilhas de 1mm, obrigado.