NSEE/LPKF #100: Vagner Possato - 15.03236-0 - 24/10/2017 - Simples

VagnerPossatoJunior, 2017-10-24:
# Sobre :

- Nome do Aluno : Vagner Possato Junior
- R.A :15.03236-0
- Habilitação : C.A
- Ano : 3
- Nome do Orientador: Alessandra
- Nome do Projeto :Portão automático
- Nome da Disciplina  :Eletrônica Digital

# PCB :

- Nome da Placa :Vagner Projeto
- Link para arquivos : https://drive.google.com/open?id=0B3CpxDT9jeGrZUpJZGF5RnZaZGs

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - (X) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (X) Simples
 - ( ) Completa

- Descrição :
 - (X) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

Renatolab, 2017-10-25:
Boa noite Vagner.
Sua placa está com uma trilha incompleta.
Favor verificar e mandar novamente.
Obrigado
Renato

VagnerPossatoJunior, 2017-10-27:
Boa noite Renato minha placa está com trilhas incompletas porque irei realizar jumpers

VagnerPossatoJunior, 2017-10-27:
Vou enviar um novo arquivo sem elas

closed 2017-10-30 by Renatolab