NSEE/LPKF #102: Felipe Brandão Ippolito - 12.01378-0 - 25/10/2017 - Simples

feippolito, 2017-10-24:
# Sobre :

- Nome do Aluno : Felipe Brandão Ippolito
- R.A : 12.01378-0
- Habilitação : Eng. Eletrônica
- Ano : 2
- Nome do Orientador: Valdir Junior
- Nome do Projeto : Fonte Linear 15V
- Nome da Disciplina  : ET3001- Projetos Eletrônicos I - Fontes de Alimentação Lineares

# PCB :

- Nome da Placa : Fonte 15 V
- Link para arquivos :  https://drive.google.com/file/d/0B5LCK_Lxpda9TG5iMUN2SzU5RGc/view

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - (X) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (X ) Simples
 - ( ) Completa

- Descrição :
 - (X) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

feippolito, 2017-10-28:
https://drive.google.com/open?id=0B5LCK_Lxpda9UVpTd3JjdzVNS1k - Novo link com o tamanho das trilhas corrigidas

Renatolab, 2017-10-30:
Felipe boa tarde
Sua Placa está pronta.
Renato

closed 2017-11-04 by Renatolab