NSEE/LPKF #103: Nemer C. - 15.03515-8 - 27/10/17 - Simples

nemerchiedde, 2017-10-27:
# Sobre :

- Nome do Aluno :  Nemer Chiedde
- R.A : 15.03515-8
- Habilitação :  Elétrica
- Ano : 3 ano
- Nome do Orientador: Sergio 
- Nome do Projeto : Placa Fonte
- Nome da Disciplina  : PAE - ET3001 - Projetos eletrônicos 1

# PCB :

- Nome da Placa : Placa Fonte
- Link para arquivos : https://www.dropbox.com/sh/535oru5qg64pbch/AADjkwkD5WrktOq5o20YSeAba?dl=0

## Informações básicas

- Pequeno descritivo sobre a placa : 
Fonte de 12V, 5V e 3V3

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - (x ) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - ( x) Simples
 - ( ) Completa

- Descrição :
 - ( x) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

nemerchiedde, 2017-10-30:
Olá,
O link que havia colocado estava incorreto, já atualizei. 
Obrigado desde já,

Nemer.

Renatolab, 2017-10-31:
Boa noite Nemer
Sua Placa está pronta
Renato

closed 2017-11-04 by Renatolab