NSEE/LPKF #104: Vagner Possato - 15.03236-0 - 27/10/2017 - Simples

VagnerPossatoJunior, 2017-10-27:
# Sobre :

Nome do Aluno : Vagner Possato Junior
R.A :15.03236-0
Habilitação : C.A
Ano : 3
Nome do Orientador: Alessandra
Nome do Projeto :Portão automático
Nome da Disciplina :Eletrônica Digital

# PCB :

Nome da Placa :Vagner Projeto
Link para arquivos : https://drive.google.com/open?id=0B3CpxDT9jeGrZUpJZGF5RnZaZGs

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 ( ) Altium

( ) Eagle

( ) Orcad

(X) UltiBoard

( ) Outro

Tipo de prototipagem

(X) Simples

( ) Completa

Descrição :

(X) Face Simples

( ) Dupla Face

( ) SMD

( ) Trilha RF

( ) Máscara

VagnerPossatoJunior, 2017-10-27:
Boa noite Renato
As trilhas que estão faltando nesse projeto são trilhas que teriam que serem feitas em duas faces, por isso pode ignora-las que irei fazer jumpers nelas, são 3 ligações.
Obrigado pela atenção

Renatolab, 2017-10-28:
Bom dia Vagner.
Sua Placa está pronta.
Renato

closed 2017-10-30 by Renatolab