NSEE/LPKF #107: César Abraham F. C. Filho - 14.00800-9 - 30/10/2017 - Simples

cesar154, 2017-10-30:
# Sobre :

- Nome do Aluno : César Abraham Flores Cisneros Filho
- R.A : 14.00800-9
- Habilitação : Eletrônica
- Ano : 3º Ano
- Nome do Orientador: Valdir Melero Junior
- Nome do Projeto : Fonte Linear Ajustável
- Nome da Disciplina  : Pae - Fontes de Alimentação Lineares

# PCB :

- Nome da Placa : Fonte Linear Ajustável
- Link para arquivos : https://drive.google.com/open?id=0B1dIjMLPlOEQaUhNU2dtbWxZd2s 

## Informações básicas

- Pequeno descritivo sobre a placa : Fonte Linear Ajustável 0-18V 

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - (X) Outro - Proteus

- Tipo de prototipagem
 - (X) Simples
 - ( ) Completa

- Descrição :
 - (X) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

Renatolab, 2017-11-04:
Bom dia César.
Seus gerbers vieram incolpletos.
Não veio o Botom Layer que são as trilhas, nem o board outline. Só chegaram os furos. O restantes dos aquivos que chegaram estão em códigos.
Aguardo contato
Obrigado
Renato

cesar154, 2017-11-04:
Bom dia,
Fiz usando Proteus então usei auxílio da internet para passar pro formato gerber, os layers não ficaram com esses nomes específicos, porém eles estão presentes, selecionando todos e abrindo, com o programa para visualizar gerber descrito no procedimento de envio, pode-se confirmar que todas os layers formam o circuito. 
Atenciosamente,
César Filho

[image]

[image]

[image]

[image]

Renatolab, 2017-11-08:
Boa tarde Cesar.
Sua Placa está pronta
Renato

closed 2017-11-09 by Renatolab