NSEE/LPKF #109: Inversor Senoidal

FeMontagneri, 2017-10-31:
# Sobre :

- Nome do Aluno : Felipe Antonio Montagneri Lucchini
- R.A : 09.00053-4
- Habilitação : Engenharia Eletrônica
- Ano : 5º
- Nome do Orientador: Alessandro Santos 
- Nome do Projeto : Inversor Senoidal
- Nome da Disciplina  : Eletrônica de Potência

# PCB :

- Nome da Placa : Inversor Senoidal
- Link para arquivos : https://www.dropbox.com/s/gpl3scx3n581ja7/Inversor%20Senoidal.rar?dl=0

## Informações básicas

- Pequeno descritivo sobre a placa : placa de potência para implementação de um inversor de onda senoidal

- Software utilizado 
 - (x) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (x) Simples
 - ( ) Completa

- Descrição :
 - (x) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - (x) Máscara 

Por favor, retirar todo o cobre que não será utilizado, pois a placa será soldada à mão.

Renatolab, 2017-11-04:
Boa tarde Felipe tudo bem?
Sua placa é mesmo duplicada? Não cabe na máquina se for as duas juntas.
Posso separar?
Obs:  As placas será feita iguais as outras , com isolação entre as trilhas. O cobre não será retirado ok? O desgaste é muito grande das brocas pra retiras o cobre e há muitas placas para serem feitas. Conto com sua compreensão.
No aguardo.
Obrigado
Renato.

FeMontagneri, 2017-11-06:
Pode separar, sem problema.

Renatolab, 2017-11-08:
Boa tarde Felipe
Não consigo separar sua placa.
Envie uma placa só, pedindo duplicidade. ok?
Aguardo
Renato

FeMontagneri, 2017-11-09:
Segue link atualizado, por favor fazer duas unidades:

https://www.dropbox.com/s/gpl3scx3n581ja7/Inversor%20Senoidal.rar?dl=0

Renatolab, 2017-11-11:
Bom dia Felipe
Você pode vir ao laboratório?

FeMontagneri, 2017-11-13:
https://www.dropbox.com/s/ip83jwkwhumspvr/Inversor%20Senoidal.rar?dl=0

FeMontagneri, 2017-11-13:
Link atualizado com arquivo de furação corrigido

Renatolab, 2017-11-14:
Boa tarde Felipe.
Só pra deixar registrado. Sua placa ficou pronta. 
Obs: Já retirada.
Renato

closed 2017-11-14 by Renatolab

Renatolab, 2017-11-21:
Felipe boa tarde.
Você fez a correções das trilas? O link ainda não veio.
Vou fazer sua placa em acrílico como havia me pedido ok?