NSEE/LPKF #111: Raphael Balista Fernandez -13.04199-0 - 13/11/2017 - Simples

RaphaelFernandez, 2017-11-09:
# Sobre :

- Nome do Aluno : Raphael Balista Fernandez
- R.A : 13.04199-0 
- Habilitação : Computação
- Ano : Quarto
- Nome do Orientador: VANDERLEI CUNHA PARRO
- Nome do Projeto : Violino
- Nome da Disciplina  : ECA411-Sistemas de Controle

# PCB :

- Nome da Placa : Placa_Cafeteira
- Link para arquivos : https://drive.google.com/drive/folders/1Q3zE2K3TKYv7vMTgSW0j0vxDAliKEN-w?usp=sharing

## Informações básicas

- Pequeno descritivo sobre a placa :Placa que sera usada para o projeto semestral de sistemas de controle

- Software utilizado 
 - ( ) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - ( X ) Outro(Protheus)

- Tipo de prototipagem
 - ( X ) Simples
 - ( ) Completa

- Descrição :
 - ( X ) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

Renatolab, 2017-11-14:
Boa tarde Raphael.
Abri seus arquivos e as trilhas estão muito finas. Não vai consegui soldar .
Tem furos que vão encostar nas trilhas.
Se possível , diminua o espaço da placa pois o circuito é relativamente pequeno.
Estou no aguardo.
Renato

RaphaelFernandez, 2017-11-17:
Ola Renato,os arquivos foram atualizados ,propus um novo layout com trilhas de 0,5mm

Renatolab, 2017-11-21:
Boa tarde Raphael.
Sua Placa está pronta.
Renato

closed 2017-11-23 by Renatolab