NSEE/LPKF #115: Carolline - 13.03045-0 - 10/04/18 - Simples

carollinehungaro, 2018-04-04:
# Sobre :

- Nome do Aluno : Carolline de Andrade Hungaro
- R.A : 13.03045-0
- Habilitação : Engenharia Elétrica
- Ano : 6 ano
- Nome do Orientador: João Carlos Lopes
- Nome do Projeto : Parkcare
- Nome da Disciplina  : TCC

# PCB :

- Nome da Placa :
- Link para arquivos : https://drive.google.com/drive/folders/1KAp3BF4N4GUrdKChK6qNKPWf0sHWI-za?usp=sharing

## Informações básicas

- Pequeno descritivo sobre a placa :

- Software utilizado 
 - (x) Altium
 - ( ) Eagle
 - ( ) Orcad
 - ( ) UltiBoard
 - ( ) Outro

- Tipo de prototipagem
 - (x) Simples
 - ( ) Completa

- Descrição :
 - (x) Face Simples
 - ( ) Dupla Face
 - ( ) SMD
 - ( ) Trilha RF
 - ( ) Máscara

Renatolab, 2018-04-09:
Boa tarde Caroline.
Você poderia comparecer ao laboratório para explicar o arquivo?
Obrigado
Renato

Renatolab, 2018-04-11:
Boa noite Caroline.
Sua placa está pronta.
Renato